whosmatt/uvmod #1: Please add these features

RE3CON, 2023-08-03:
First of all many thanks for your great work! keep it up.
It would be great to add:
- AM RX and AM TX on all Frequencies
- Frequency Steps enhanced e.g. 8.33khz, 500khz, 0.01khz, 1,25khz, 10000khz,...
- 18-1300 MHz RX/TX already added?

An Android OTG Firmware Flasher as by modded Firmwares for MD380 would be great...
73's

whosmatt, 2023-08-04:
Thank you for the feedback. 
 
 - AM on all frequencies seems to be buggy right now (breaks channel lists)
 - additional freq steps is flawed due to an overloaded constant, but the regular freq steps will be added soon
 - 18 - 999MHz RX is added, 1300 needs another mod to be displayable on the screen
 - making TX unlock mods easily accessible is not a good idea since the stock radio hardware is near useless outside UHF/VHF, with illegal spurious emissions. 
 - This tool is specifically designed for the UV-K5 based radios, since they all use the same firmware. The MD380 has its own firmware and lacks the modding scene right now.

RE3CON, 2023-08-04:
Hi
going over 999MHz does not more fit on display in one line and it looks like 000,123 etc... so that the highest keypad input is 999.xxxx and frequencies can not more with keypad above 999MHz applied. The only way is with bw steps to go higher. Not sure if it really receive above 800 or its just like a mirror from lower frequ. The same with TX it works more or less fine from 50-899 MHz. On some frequ. the power output is on freq x2 same or higher. For example TX on 60 MHz transmite as well 60mhz x2 = 120 MHz almost higher as on the target freq. I think by the old MD380 projects will not more come anything new cause 3 from 4 different firmware mods are inactive since 2019 the remaining only releasing bugfixes. The md380 firmware chip have way more mem space as the 60kb by these new analog radios. 
Appreciate your work. Its really great.

closed 2023-08-04 by whosmatt

whosmatt, 2023-08-04:
> The same with TX it works more or less fine from 50-899 MHz.
This is completely untrue, transmission power goes down to microwatts on the carrier frequency if you go far away from the design frequency. Meanwhile plenty of harmonics show up as spurious emissions, sometimes stronger than the carrier, leaking into frequencies illegal in most countries, and seriously dangerous when interfering with sensitive communications such as aerospace. Your example of TX on 60MHz will leak into 120MHz, 180MHz, 240MHz and some more. 120MHz happens to be in the worldwide air band. 
**Unlocking TX on other bands without an external filter or serious hardware modification is reckless.** 

RX is usable on most frequencies but sensitivity gets very poor towards the bottom and top of the spectrum. Some people are working on a mod to display frequencies above 999MHz, I will implement it when its done but I prefer to not have buggy mods in the meantime.

reppad, 2023-08-04:
>     * AM on all frequencies seems to be buggy right now (breaks channel lists)
I believe you are referring to the mod consisting of extending the F2 band like [this firmware](https://github.com/Tunas1337/UV-K5-Modded-Firmwares/blob/main/k5_v2.01.19_mod_AM_18to850mhz.md).
From what I understand [this new mod](https://github.com/amnemonic/Quansheng_UV-K5_Firmware/blob/main/uvmod_kitchen/mod_enable_am_everywhere.py) works differently and allows AM to be activated on all bands.
I have not tested but I assume that It shouldn't cause the same problem ?

RE3CON, 2023-08-04:
You are right AM TX mod from piotr022 does actually not work. I compiled it and push to activate Amplitute Modulation  TX the flashlight 2nd side button. 

The AM RX mod works perfect.
Great that you have frequency steps and other working stuff added.

RE3CON, 2023-08-06:
nope I really do mean TX (transmite) not receive in AM instead of FM but looking at the hardware chip specs it may not work but who knows. I [compiled the am tx](https://github.com/RE3CON/UV_K5_playground/releases) mod [from piotr022](https://github.com/piotr022/UV_K5_playground/tree/main/src)
It replace the flashlight function with an extra menu similar as by rssi_sbar_hot mod
by sbar_hot you got a menu on flashlight button where AM RX receive can be set between 0/1/2
[image]

commit 45dad33b (2023-08-04): added frequency steps mod, closes #1