whosmatt/uvmod #2: Dodatkowe kroki strojenia i podmiana kanałów NOAA

Pomidor78, 2023-08-06:
1. Można dorzucić dodatkowe kroki strojenia (dzięki czemu czas podświetlenia może być przedłużony do 10 sekund) - sprawdzone, działa razem z tym modem
2. Podmiana kanałów NOAA na inne (ja wpisuję tam kanały FM link bo tylko te wg. mnie nadają się do słuchania z otwartym SQ) -sprawdzone, działa razem z tym modem
Czy jeśli można ustawić częstotliwość roger beep na 1050 Hz to można nim odblokować odbiór na kanałach NOAA ?

I jeszcze raz wielkie dzięki za UVMOD, chciałem coś takiego zrobić w Python ale ten online jest o wiele lepszy!
[image]

whosmatt, 2023-08-06:
Additional tuning steps is currently broken because of an overloaded variable. Besides backlight, other menu variables also go higher, and they can crash the radio. Use the "Double backlight duration" mod instead. 

I will add the NOAA frequencies mod soon. 

Please do not transmit on NOAA emergency broadcast channels.

Pomidor78, 2023-08-06:
Sprawdziłem, ustawiając roger beep na 1050 Hz da się otwierać kanały PMR wpisane zamiast kanałów NOAA
[image]
[image]

closed 2023-08-06 by whosmatt