whosmatt/uvmod #6: 18~1300 TX enabled after using RSSI

troilus, 2023-08-09:


whosmatt, 2023-08-09:
I did some testing

Stock should be
F1: 50.0000-76.0000MHz (RX only)
F2: 108-.0000-135.9750MHz (RX only)
F3: 136.0000-173.9750MHz (TX and RX both)
F4: 174.0000-349.9750MHz (RX only)
F5: 350.0000-399.9750MHz (RX only)
F6: 400.0000-469.9750MHz (TX and RX both)
F7: 470.0000-600.0000MHz (RX only)

But with RSSI firmware I got
F1: 50.0000-76.0000MHz (RX only) No TX
F2: 108-.0000-135.9750MHz (RX only) No TX
F3: 136.0000-173.9750MHz (TX and RX both) TX
F4: 174.0000-349.9750MHz (RX only) **TX**
F5: 350.0000-399.9750MHz (RX only) **TX**
F6: 400.0000-469.9750MHz (TX and RX both) TX
F7: 470.0000-600.0000MHz (RX only) **TX**

It seems there is indeed an issue with piotr's RSSI firmware.

Pomidor78, 2023-08-13:
Tak, trzeba wejść w menu dodatkowe i wyłączyć nadawanie w tych pasmach.
[image]

analystcmyk, 2023-08-16:
Translated : **_Yes, you need to go into the additional menu and disable broadcasting in these bands._**

Related : Can someone confirm that after applying the mod the F-LOCK ( e.g. CE ) is not working anymore? 

I like having the RSSI mod but not losing the F-LOCK option..

Suggest to add a mod similar to 
```
Mod_DisableTX 
Mod_EnableTXEverywhereButAirBand

```
but called for example 
```
Mod_DisableTX-FLock-CE
Mod_DisableTX-FLock-FCC
Mod_DisableTX-FLock-GB
Mod_DisableTX-FLock-430
Mod_DisableTX-FLock-438
```

This makes it easier for people to not go wrong by mistake.

whosmatt, 2023-08-16:
I couldnt figure out the root cause yet but I did add a warning in https://github.com/whosmatt/uvmod/commit/824e2f58ad7f40a41ca99651fe2127acdfddd473

Pomidor78, 2023-08-18:
Możesz dopisać że nadawanie można wyłączyć w dodatkowym menu

whosmatt, 2023-08-18:
The problem is that the mod enables TX without me changing any settings.

Pomidor78, 2023-08-19:
Tak jakby aktywowały się domyślne ustawienia tego firmware. Tak jak w nowym radio. Akurat tu nie jest to duży problem bo można to wyłączyć