whosmatt/uvmod #9: can you implement this mod ? ZVEI2 5-tone

Braccato, 2023-08-21:
can you implement this mod ? ZVEI2 5-tone

whosmatt, 2023-08-21:
sure, how exactly would you describe this mod? what does it replace and what does it do?

Braccato, 2023-08-24:
puoi implementare questa mod? ZVEI2 5 toni

whosmatt, 2023-08-24:
right now the mod seems a little too hacky for my liking, many steps involved to make it work. ill look into options to make it simpler

Braccato, 2023-08-25:
implement a tone like this:

https://github.com/whosmatt/uvmod/assets/141162772/95cb2ab8-303b-4571-9d9c-14cfb1f8ac2b

SylarMa, 2023-09-17:
I want to use it

How can I find this mod?

You down code 123a1

Why does Motorola display 13921?

@Braccato

SylarMa, 2023-09-17:
I want to use it

How can I find this firmware?

You down code 123a1

Why does Motorola display 13921?

@Braccato

Braccato, 2023-09-17:
Here: http://uvmod.ddns.net/

SylarMa, 2023-09-17:
https://github.com/whosmatt/uvmod/issues/9#issuecomment-1722503017

thank you

SylarMa, 2023-09-17:
cool