whosmatt/uvmod #13: can you make it able to transmit in FM radio signals?

salmannn5, 2023-08-26:
As a tour guide, i need an FM transmitter (88 MHz to 108MHz). at the same time i need a walkie talkie too.
Can you make it possible this UV-K5 to transmit in FM Radio signals, and can be received clearly by regular radio receiver?
Thank you

OneOfEleven, 2023-08-26:
The radios TX chain has been designed to transmit only within the 136-174 420-470'ish range, transmitting outside of those areas has found to be a bad idea due to the large amplitude spurious carriers it produces when you try - do so as your own risk (country laws might get you found out).

Anyway, not sure if the BX radio chip will produce the level of FM deviation required by 88-108 WBFM receivers.

OneOfEleven, 2023-08-26:
I can see this radio being band in certain countries (with certainty) as is (very soon too), mainly due to the air-band limit being removed - it's been a seriously bad idea to have given the morons out there the ability to transmit across the air-band (118-136), and also the higher military band.

Am glad all countries have air-band DF'ing all built in (to instantly locate lost aircraft), and so making it easy to pin point the dummies that are transmitting in the air bands(s).

whosmatt, 2023-08-26:
Check the instructions on UVMOD to see the hidden mods. For the same reason
you stated I decided to hide TX unlock mods behind a little bit of warning
and info.

whosmatt, 2023-08-26:
The K5 can not really transmit on FM radio frequencies. It uses a second chip for FM radio which is receiver only. The main chip could transmit and receive on those freqs too but the bandwidth is far too low to be usable, I couldnt get it to demodulate and if you were to send on FM it would just be uselessly quiet. It would also be a very dirty transmission and way too much power, silencing many radios in a large area. Very likely illegal in your country too.

closed 2023-08-26 by whosmatt

RE3CON, 2023-08-26:
for fm radio the frequency modulation fm is to small with max. 24khz which
is with this hardware possible. It needs different fm modulator circuide
with wwfm for a greater fm hub spectrum results to be way louder than these
walkie talkies can do this. You are looking for fm radio transmitters in
size of a stand alone microphone with build in fm tramsmitter mostly
without a fixed freq crystal made but very cheap to found on the market for
range FM 88-108MHz. Maybe 8 to 12 bugs.