whosmatt/uvmod #14: Implementation of two mods

dammh, 2023-08-26:
Hi and thank you for your job!

Is it possible to implement those two mods:
mod_custom_freq_ranges.py
mod_more_freq_steps_and_backlight_duration.py

both are from uvmod_kitchen

Thank you!

whosmatt, 2023-08-27:
custom freq ranges is already implemented. 

more freq ranges is broken and there is currently no good fix. when using this mod, some menu options will crash or freeze the radio or corrupt eeprom, so i avoided adding it for now.

dammh, 2023-08-27:
Oh I see. Thank you anyway. I hope a fix will come soon.
About custom freq ranges. I don't see it in the UVMOD page. I want to be able to modify the frequency range of each bands.
I can work the other way around and use uvmod_kitchen but there I can't find a way to do Custom TX Range.

whosmatt, 2023-08-27:
oh thats what you mean, ill implement that next week. 
you can always prepare a bin in uvmod kitchen and then use it in uvmod to add additional mods. it will work as long as you dont use conflicting mods.

whosmatt, 2023-08-29:
> Oh I see. Thank you anyway. I hope a fix will come soon. About custom freq ranges. I don't see it in the UVMOD page. I want to be able to modify the frequency range of each bands. I can work the other way around and use uvmod_kitchen but there I can't find a way to do Custom TX Range.

custom tx range is difficult because there are multiple checks involved rather than a simple frequency table. there is a mod called CustomTXRange which replaces all the checks with a simplified one-range function, that's all we got at the moment. If you need to enable more TX bands, check the secret menu available by holding PTT and macro key 1 during powerup. 

mod_custom_freq_ranges is now integrated into uvmod. you can find it as Custom Frequency Ranges.

closed 2023-08-29 by whosmatt