whosmatt/uvmod #15: Problem with scanning

PolokyCZ, 2023-08-28:
Scanning problem. When I start scanning, the display light is steady and won't turn off. Would it be better when I turn on scanning to have it glow for 5s, the display would go off and when a signal is detected it would light up again? Would you write the mode please?

whosmatt, 2023-08-29:
Right now with mods this is pretty difficult, but we are almost finished with a complete rewrite of the firmware. Once thats complete, we will make a improved and smaller version of the firmware that has this as an improvement.

closed 2023-08-29 by whosmatt

PolokyCZ, 2023-10-03:
Hello. How's it going with the firmware update?

whosmatt, 2023-10-03:
open source firmware is already available in a couple different versions  
https://github.com/DualTachyon/uv-k5-firmware

take a look in the readme for links to customized variants. 
the firmware.packed.bin files from these firmwares can be flashed with uvmod, but you cannot use any mods. 

a universal uvmod version is on the way but i cant finish it without help, the old mods need to be reimplemented in the new format