whosmatt/uvmod #16: Web flasher does not work on Linux / Chromium

AbortRetryFail, 2023-08-28:
The UVMOD "Flash directly" button fails on my Linux machines. I've tested with Chromium 116 on Arch Linux and Chromium 115 on Debian Sid.

I'm using a FTDI FT232 based programming cable that I have both successfully used on Linux with CHIRP to program the UV-K5 and other radios, as well as to flash the UV-K5 using UVMOD's "Flash directly" feature on a Windows 10 machine using Edge. The cable also works with k5prog. The cable works and so does Linux's access to the serial port.

It doesn't provide any useful error messages to troubleshoot the problem:
"No data received, is the radio connected and in flash mode? Please try again."

AbortRetryFail, 2023-08-28:
Yep, the radio is in bootloader mode with the screen blank and the flashlight LED on. This programming cable also has red and green LEDs for TX and RX on the USB side. Under normal operation you can see both flashing (as it does with CHIRP and using UVMOD on Windows) - On Linux w/Chromium I only see the red one flash 8 times. If I have to guess, whatever it is trying to send is being mangled in some way the radio is not expecting, and not responding.

Nothing shows up in the DevTools console.

whosmatt, 2023-08-29:
Nothing in the dev console means that it does not decode any packets. I have a hunch that something in your browser is not working or working differently related to webserial. perhaps it is not setting the baudrate correctly. ill give you a modified qsflash.js later which will print more debug output. while in flash mode, the radio should constantly spam packets.

whosmatt, 2023-08-29:
With todays update, the flasher will now log the entire buffer to the dev console whenever that error message shows. I already noticed that it fails to parse packets on the first attempt after reloading the page, at least for me. But I think this is unrelated to your issue. 
If you could reproduce your issue and send me the dev console output, that would be helpful.

AbortRetryFail, 2023-09-03:
Well, either you fixed it, or something else was wrong, because it works now.

Sorry for the delayed response, it's been a week.

closed 2023-09-03 by AbortRetryFail