whosmatt/uvmod #18: Question? AM TX  is possible?

kinetikFR, 2023-09-02:
i would like know  it is possible AM TX on Aviation range

whosmatt, 2023-09-02:
possible yes, already implemented too but HIGHLY dangerous and illegal if youre planning to transmit without a dummy load.

kinetikFR, 2023-09-03:
How do I proceed, it would only be for study purposes I recently bought a hackrf and was testing it

whosmatt, 2023-09-03:
Which attenuator are you using?

whosmatt, 2023-09-03:
Besides, for AM TX you need to modify the hardware of the radio, you would need to inject baseband into the pa to modulate it. And please remember that transmitting on airband is not just very illegal, most countries do have a lost aircraft detection system, so they get an alert with your approximate location if you do.

closed 2023-09-04 by whosmatt