whosmatt/uvmod #20: TRANSIMITIR AIR BAND EN MODO AM y no FM

mangrullo11, 2023-09-06:
Hola! primero quiero felicitarlos por el trabajo que hacen! 
Mi nombre es Roberto Garcia, soy radioaficionado (LU1HCM) también estoy en el ambiente aeronáutico, soy miembro de la comisión directiva del aeroclub de mi ciudad, y estuvimos viendo con algunos pilotos la posibilidad de tener algunos handy para el uso en tierra y comprar uno de banda aeronáutica resulta muy caro en Argentina... vi la posibilidad de usar uno de estos Quansheng y compre uno, ya le cambie el firmware y si bien ahora podemos transmitir en la banda aeronáutica, el problema es que la transmisión es en modo FM y no AM como debería ser... hay posibilidad de que eso se pueda cambiar por software o es una limitación de hardware?

lohtse, 2023-09-07:
If you are and Actual HAM operator and involved in aeronautical enro   I don't give a damn what you think about using the K5 for Airband coms. DON'T use it one you will likely DAMAGE you TWO and the most important you will BLEED through everyone else's Channels due to poor harmonics that will endanger other people, Plus as well as incredibly poor Transmit signal as it is NOT a TRUE AM TX/RX HT as it DOES NOT have an actual AM chip inside... ALL this info is available throught the Groups and internet..

And before you even try and say I have no idea, 

Qualfied pilot for 30+ years 
My current aircraft
Robison's r66
Eurocopter H175
DHC2 Beaver
DHC3 Otter
Antonov AN-2

So yeah I know a thing or two and a K5 would only ever be used it that was a last resort in an emergency, But NEVER as my actual coms....  You want Airband coms get a proper airband set thats the best only only option.

whosmatt, 2023-09-07:
Have to fully agree with lohtse. 
The sideband emissions and horrible quality make the quansheng radios completely unsuitable for critical infrastructure such as airband. 

Even for emergency purposes, the terrible sensitivity and AM demodulation make it basically useless. AM RX is a dirty software hack, leading to extremely noisy and distorted reception. 
AM TX with the microphone is not possible with this hardware.

closed 2023-09-07 by whosmatt