whosmatt/uvmod #21: Please update fagci to v 78

LanWolf, 2023-09-09:
Could you please update the fagci mod to  v78? This fixes the stepping on 8.33 for cleaner reception.
Thank you!

whosmatt, 2023-09-12:
i am preparing a major update, which may take some more time. apps will be updated in the process too.

TobiasPSP, 2023-10-19:
Great news. You have done an excellent state of art job with the web patcher and direct upload - kudos! 

On the app side, a lot of progress has been made so I too would love to see an update to the most recent app versions when you find the time. 

Again, many thanks for your awesome work!

whosmatt, 2023-10-19:
The project is on hold until more contributors are found. The new version is ready but the mods need to be individually ported to the new Multi-Version format and I dont have the time to do it all myself

TobiasPSP, 2023-10-19:
> The project is on hold until more contributors are found. The new version is ready but the mods need to be individually ported to the new Multi-Version format and I dont have the time to do it all myself

Fully understood. 

I wish I could help but that’s beyond my scope.

Thank you for your work. You already made my day. Keeping fingers crossed that eventually we see more contributors.

whosmatt, 2023-10-19:
Yeah, documentation here is sparse. Most of the info is floating around the telegram development groups and some of it is compiled in the UV K5 wiki. You can find it browsing the links from my readme. 

Bricking is virtually impossible since UVMOD does not flash over the bootloader, so a radio can always be flashed back when something does not boot. A maliciously crafted firmware could brick the radio though. 

The current live UVMOD version uses hardcoded v26 offsets and various forks exist for newer versions. So with my version only v26 firmware with unchanged offsets works, anything else will produce gibberish that may not boot and may malfunction. 

The dev version uses a symbol table file along with the firmware file and dynamically loads the necessary offsets. That's where I need help porting the mods though, I can't find the time to do it myself. 
The dev version has working Multi-Version and multi-language support but currently only one mod. It is not available online but in this repo on the dev branch. 
It also includes a half finished configurator (CPS Software replacement) tool with a v26 tx power bug detector.