whosmatt/uvmod #24: TX unlock, have frto. appears tx carrier but no audio on airband. tried mic gain no change? other freq. ok.

DonaldNZ, 2023-10-04:


whosmatt, 2023-10-05:
please provide more info

DonaldNZ, 2023-10-05:
Thank you.

Uv-5r plus
Using uvmod
No other prior firmware updates.
I selected tx everywhere.
On tx airband no audio transmitted, like mic disabled. Sounds like carrier is tx,
 as strong signal rx'd into another airband ht.
How should I make the mic work please?

If I select 144.250 fm mic ok
Also battery percentage & RSSI 
not included although selected in same update.

whosmatt, 2023-10-05:
Keep in mind that the radio can only transmit FM. It is unsuitable for air
band use due to the poor quality of transmission.

DonaldNZ, 2023-10-05:
The carrier sounds clear and strong to fully and cleanly open the rx squelch without any static

whosmatt, 2023-10-05:
Are you trying to receive AM?

DonaldNZ, 2023-10-05:
Both receive and transmit. I'll accept the low quality. I have a ham friend
who may be able to help out with a filter in the aerial and has the
equipment to scan for spurious harmonics. I have a frto ticket [ Flight
radio telephone operator ], so I can and do legally transmit on airband.

OneOfEleven, 2023-10-05:
The K5/K6 is sold for ham radio use (see QS's website), that means they don't need to be type approved, which saves the manufacturer/sellers a lot of money.

Airband/Commercial transceivers are all type approved, and no doubt airband transceivers have stricter standards than standard commercial type approval, which means ham radios are not allowed to be used on the airbands for transmission, especially when they transmit FM rather than AM. AM requires linear TX PA's, FM doesn't.

But then, as in all parts of life, everyone has to make their own mind up/choices about the things do in life.

closed 2024-11-01 by whosmatt