whosmatt/uvmod #26: AM audio fix by OneOfEleven

krotoniec, 2023-10-21:
Firs of all, great work, it transforms cheap radios into something else, I truly enjoy it :)
There is some sort of an issue with AM audio quality in this radios, although OneOfEleven seems to have fix that's working.
https://github.com/OneOfEleven/uv-k5-firmware-custom
Any chances it could be implemented in UVMOD website?

whosmatt, 2023-10-22:
the new dev version is automatically compatible with all oefw forks, including 1o11. 
mods need to be ported though and I can't do it alone, I need contributors on this project. 

that being said, this radio will always be crap for AM. 1o11 is just a slight improvement. AM is a dirty software hack because the hardware does not support it.

Cmed1967, 2023-10-22:
Really hoping the UVMod website can be updated to allow for version 27
firmware as well.

closed 2023-10-23 by krotoniec