whosmatt/uvmod #31: Flashing with Ubuntu 23.10 & Chrome 119.0.6045.159 doesn't work

mpyllan, 2023-11-20:
I tried  to flash a k5(8) but i get a  "No data received, is the radio connected and in flash mode? Please try again." error. Don't know how to keep going. In any case thank you for your work, it's amazing.

The radio is in bootloader mode and connected according to the instructions using the appropriate order.

Here's my conf:

OS
PRETTY_NAME="Ubuntu 23.10"
NAME="Ubuntu"
VERSION_ID="23.10"
VERSION="23.10 (Mantic Minotaur)"
VERSION_CODENAME=mantic
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=mantic
LOGO=ubuntu-logo

Web Browser
Chrome Version 119.0.6045.159 (Official Build) unknown (64-bit)

I attached two screenshots of the 1, the console and 2 the interface. The plain text is below the pics.
[image]
[image]

The browser console shows the following data:

Current Buffer: Uint8Array(164) [
171, 150, 156, 16, 53, 248, 88, 109, 171, 35, 171, 205, 36, 0, 14, 105, 147, 
222, 105, 246, 75, 45, 102, 50, 87, 105, 89, 131, 140, 75, 153, 28, 191, 189, 
61, 5, 227, 64, 82, 40, 15, 22, 108, 145, 220, 255, 220, 186, 171, 205, 164, 
14, 105, 52, 230, 105, 118, 75, 45, 230, 50, 87, 105, 89, 131, 140, 108, 98, 
230, 28, 235, 98, 112, 15, 5, 227, 157, 9, 233, 128, 150, 9, 139, 100, 255, 
255, 167, 255, 171, 205, 18, 14, 105, 52, 230, 105, 105, 246, 75, 45, …]

Current Buffer (other format):
{
    "0": 171,
    "1": 150,
    "2": 156,
    "3": 16,
    "4": 53,
    "5": 248,
    "6": 88,
    "7": 109,
    "8": 171,
    "9": 35,
    "10": 171,
    "11": 205,
    "12": 36,
    "13": 0,
    "14": 14,
    "15": 105,
    "16": 147,
    "17": 222,
    "18": 105,
    "19": 246,
    "20": 75,
    "21": 45,
    "22": 102,
    "23": 50,
    "24": 87,
    "25": 105,
    "26": 89,
    "27": 131,
    "28": 140,
    "29": 75,
    "30": 153,
    "31": 28,
    "32": 191,
    "33": 189,
    "34": 61,
    "35": 5,
    "36": 227,
    "37": 64,
    "38": 82,
    "39": 40,
    "40": 15,
    "41": 22,
    "42": 108,
    "43": 145,
    "44": 220,
    "45": 255,
    "46": 220,
    "47": 186,
    "48": 171,
    "49": 205,
    "50": 164,
    "51": 14,
    "52": 105,
    "53": 52,
    "54": 230,
    "55": 105,
    "56": 118,
    "57": 75,
    "58": 45,
    "59": 230,
    "60": 50,
    "61": 87,
    "62": 105,
    "63": 89,
    "64": 131,
    "65": 140,
    "66": 108,
    "67": 98,
    "68": 230,
    "69": 28,
    "70": 235,
    "71": 98,
    "72": 112,
    "73": 15,
    "74": 5,
    "75": 227,
    "76": 157,
    "77": 9,
    "78": 233,
    "79": 128,
    "80": 150,
    "81": 9,
    "82": 139,
    "83": 100,
    "84": 255,
    "85": 255,
    "86": 167,
    "87": 255,
    "88": 171,
    "89": 205,
    "90": 18,
    "91": 14,
    "92": 105,
    "93": 52,
    "94": 230,
    "95": 105,
    "96": 105,
    "97": 246,
    "98": 75,
    "99": 45,
    "100": 214,
    "101": 198,
    "102": 5,
    "103": 105,
    "104": 89,
    "105": 131,
    "106": 140,
    "107": 108,
    "108": 98,
    "109": 230,
    "110": 244,
    "111": 235,
    "112": 130,
    "113": 61,
    "114": 133,
    "115": 60,
    "116": 157,
    "117": 9,
    "118": 233,
    "119": 128,
    "120": 17,
    "121": 139,
    "122": 100,
    "123": 255,
    "124": 220,
    "125": 186,
    "126": 171,
    "127": 205,
    "128": 36,
    "129": 0,
    "130": 40,
    "131": 139,
    "132": 50,
    "133": 47,
    "134": 147,
    "135": 104,
    "136": 170,
    "137": 73,
    "138": 77,
    "139": 116,
    "140": 80,
    "141": 43,
    "142": 209,
    "143": 140,
    "144": 108,
    "145": 226,
    "146": 30,
    "147": 244,
    "148": 235,
    "149": 130,
    "150": 61,
    "151": 5,
    "152": 227,
    "153": 64,
    "154": 39,
    "155": 9,
    "156": 122,
    "157": 22,
    "158": 108,
    "159": 20,
    "160": 198,
    "161": 255,
    "162": 220,
    "163": 186
}

ab969c1035f8586dab23abcd24000e6993de69f64b2d6632576959838c4b991cbfbd3d05e34052280f166c91dcffdcbaabcd
a40e6934e669764b2de632576959838c6c62e61ceb62700f05e39d09e98096098b64ffffa7ffabcd120e6934e66969f64b2d
d6c6056959838c6c62e6f4eb823d853c9d09e980118b64ffdcbaabcd2400288b322f9368aa494d74502bd18c6ce21ef4eb82
3d05e34027097a166c14c6ffdcba

whosmatt, 2023-11-20:
Not sure what's going on here. I assume flashing with the original tool (or
k5prog) works with this particular radio?

mpyllan, 2023-11-21:
Apparently, k5prog reads the EEPROM (attached) using ./k5prog -r -v

   read block 0x1F80  98%
   Sucessfuly read eeprom

[EEPROM.txt](https://github.com/whosmatt/uvmod/files/13425126/EEPROM.txt)

I don't have access to a Windows box now (working on it). 
I'm trying to find a semi-secure way to test writing firmware using k5prog.

whosmatt, 2023-11-21:
If k5prog can read eeprom, the radio is not in bootloader mode. Try using the flash option in k5prog to flash the radio, same setup as uvmod or the windows tool. Either way, in bootloader mode the display should be off and the flashlight on.

mpyllan, 2023-11-21:
There must be something wrong with this particular device.  Apparently, the radio is in bootloader mode (light up, screen off after PTT+ON), but it still reads the EEPROM.  Cannot flash k5_flash.raw as described in the k5prog repo. Tried with CHIRP next, but didn't work either.  The cable works with other radio, and CHIRP, so I assume this device has something peculiar. I'll try with Windows later.

In any case, the problem has nothing to do with your software. Sorry for the inconvenience. Thank you very much.

closed 2023-11-21 by mpyllan

whosmatt, 2023-11-21:
the windows software can tell you the firmware version of the radio. it would be interesting to see if this is related to some newer revision of the bootloader and firmware.