whosmatt/uvmod #33: Mod for listening to satellites with doppler effect

kaushleshchandel, 2023-11-28:
Hi, I like the project and want to create a mod. I am programmer and Ill start working on it.

To listen to satellites, you need to keep switching frequencies every minute, due to doppler effect. I am hoping to create a mod that lets you select a satellite, and its frequency. Then the radio automatically changes frequency every minute. 

Is there anything like or similar to that? or a starting point that I can use to start programming something for this?

Thanks for the good work you are doing on the mods!!!

kaushleshchandel, 2023-11-28:
I am not sure if its possible, but can we change frequencies over serial port?

troilus, 2023-11-29:
https://www.bilibili.com/video/BV1H94y1G7Xf/

whosmatt, 2023-11-29:
certainly possible but you will have to find someone else who is willing to put in the time to develop that

closed 2023-11-29 by whosmatt

whosmatt, 2023-11-29:
> Thanks Matt.
>
> I am a programmer. I have built a product for ham radio, called Boondock
> Echo. Check it out when you get time. It’s a ham radio recording, playback
> and transmit device. We are active with hackaday and we won 4th prize last
> year. We have built it as a product and about to run a crowd supply
> campaign.
>
> http://boondockecho.com
>
>
> I want to work on the feature. If you can point me in the right direction
> to get started, that would be great.
>
> Thanks,
>
> KC

In that case, just check out the various OEFW (open edition firmware) repos
for the uv-k5. They contain a full C reimplementation of the original
firmware and it is relatively easy to add arbitrary new features. I would
suggest looking at the serial code and implementing the frequency set
feature as an additional command. I believe there are also unused serial
commands that can have this functionality added.
https://github.com/DualTachyon/uv-k5-firmware