whosmatt/uvmod #37: Unable to flash mods over Egzumer v0.22

kyebrown, 2024-02-09:
I'm having trouble flashing additional mods (custom boot screen and 'Futuristic Font') with Egzumer v0.22, even when the Egzumer .bin is uploaded as the firmware version. 
I have a UV-K6 and have selected this in the menu. 

When I turn my radio on after the patch has completed, the boot screen hasn't changed and the font hasn't either - but the font mod does cause a few random parentheses to show up in various places on the screen. 

Is it even possible to flash these mods alongside Egzumer?

Apologies if this question is a silly one, I'm pretty inexperienced with this stuff.

kyebrown, 2024-02-09:
> Mods are only meant for stock fw's not custom....

Thanks for the quick response. I did think this might be the case, but as there was an option to select/upload the FW version I wasn't 100% sure. Like I said, bit of a noob to this lol.

closed 2024-02-09 by kyebrown

whosmatt, 2024-02-09:
yeah, the current version of uvmod only supports stock firmware