whosmatt/uvmod #45: Latest Mac OS unable to communicate via chrome

imzenreally, 2024-11-18:
I've googled thinking I'm missing something and have come up dry 

But I can get chirp to talk to the radio but the web flasher refuses to recognize the radio (using the same serial port etc)

Did something change - I can't believe I'm the only donk using the latest Mac OS

whosmatt, 2024-11-18:
im assuming this is #44 

does the flasher start flashing and then time out? you can see more about the status by opening the browser console and checking the output

bjobr, 2024-12-29:
I have the same problem on MacOS Sequoia and Chrome.
Output in flasher:
```
Connecting to the serial port...
No data received, is the radio connected and in flash mode? Please try again.

```
Console only shows this:
[image]

bjobr, 2024-12-29:
Possibly related: https://discussions.apple.com/thread/255763367?sortBy=rank